SiEPIC/SiEPIC-Tools #4: Monte Carlo GUI needs implementation

bpoulse, 2017-12-14:
Actual form is implemented using ui file. Needs logic.

For MC script, only get_parameters etc must be called

lukasc-ubc, 2018-01-14:
Hi Brett,

Are you working on this one, or can we just use the previous code to create the GUI?

thanks
Lukas

bpoulse, 2018-01-14:
Hey Lukas,

It was partially implemented, but I left it unfinished. I have finished it to the point where the GUI is complete. The code is implemented in core.py and lumerical.py but nothing is launched after the window is closed. `MC_GUI.get_parameters()` returns the following object structure:
```python
{
  'num_wafers': Number of wafers
  'num_dies': Number of dies
  'technology': Chosen technology
  'histograms': {
    'fsp': Boolean for generating free spectral range histogram
    'gain': Boolean for generating gain histogram
    'wavelength': Boolean for generating wavelength histogram
  },
  'waf_var': {
    'width': {
      'std_dev': Width standard deviation within a wafer
      'corr_len': Width correlation length within a wafer
    },
    'height': {
      'std_dev': Height standard deviation within a wafer
      'corr_len': Height correlation length within a wafer
    }
  },
  'waf_to_waf_var': {
    'width': {
      'std_dev': Width standard deviation from wafer to wafer
    },
    'thickness': {
      'std_dev': Thickness standard deviation from wafer to wafer
    }
  }
}
```
But I have not filled in defaults for the different technologies.

Additionally, I spent some time on the GUI's, I believe they are actually fixed this time... the window doesn't always seem to size properly given different screens/operating systems, but it can now be resized like a normal window so everything is visible.

lukasc-ubc, 2018-01-15:
Reverted the last commit since it broke it for OSX / 0.24.

closed 2018-02-08 by lukasc-ubc