SiEPIC/SiEPIC-Tools #22: py vs lym

lukasc-ubc, 2017-12-15:
I understand the use of py files, since you can import them. But the debugger and "singleton-like" behaviour means we need to exit KLayout each time changes are made to py files.

Can we minimize the use of py files?

Lym files work well in the debugger where you can see the error, breakpoints, etc.

bpoulse, 2017-12-15:
The way I worked around this was having a test lym file, 'SiEPIC - Test Environment.lym'
Any functions I wanted to work on, I would work on in this file. For instance
```
def arc_test():
  return [pya.Point(0,0)]

from SiEPIC import utils
utils.arc = arc_test
```
replaces the original arc function in utils with the new one I wrote. The changes carry through since modules are singleton like.

bpoulse, 2017-12-16:
Added force reload of SiEPIC library if already initialized. Rerunning SiEPIC.lym reloads module.

closed 2017-12-16 by bpoulse