SiEPIC/SiEPIC-Tools #25: Windows GUI font scaling issue

lukasc-ubc, 2017-12-16:
On Windows, when I change the display configuration "change the size of all items", e.g., 200%, the GUI for Waveguides gets messed up.

All the other KLayout GUIs are fine.

[image]

bpoulse, 2017-12-17:
I tried a fix for the gui, can you let me know what the result on that VM is?

lukasc-ubc, 2017-12-17:
[image]

bpoulse, 2017-12-22:
@lukasc-ubc can you once again let me know if the changes I made make any difference at all? It works with changing my monitor resolution but not sure if that is the issue.

lukasc-ubc, 2017-12-23:
Looks great on OSX:
[image]

lukasc-ubc, 2017-12-23:
Here is what it looks like on Windows (7 rr 8) with scaling set to 150%:

[image]

lukasc-ubc, 2017-12-23:
Is there any way to make the stretch thing on the bottom right corner enabled, so that we can stretch the window if necessary?

bpoulse, 2017-12-23:
It should be enabled already. I will tinker around some more.

closed 2018-02-08 by lukasc-ubc