SiEPIC/SiEPIC-Tools #27: Hot keys

lukasc-ubc, 2017-12-16:
I don't what changed.  But yesterday the hotkeys were working, and were automatically loaded.

Sometime tonight they stopped.  Have to run manually "SiEPIC installation and initalization", each time a new layout is created or a new GDS file is loaded.

lukasc-ubc, 2017-12-16:
Actually, I checked the version before I made the edits tonight, and it seems like this was also an issue before too...

bpoulse, 2017-12-16:
There seem to be several issues with 0.25 involving this. For myself, the hotkeys work with no issue on Windows 10. However, your addition of the 'SiEPIC.lym' to the menu does not show up (as mentioned before) regardless of if it is in the salt folder or local.

lukasc-ubc, 2017-12-16:
Could you post something on KLayout GitHub?

I believe the old PDK works though right?

Lukas Chrostowski
Professor, ECE
University of British Columbia
604-822-8507
http://www.mina.ubc.ca/lukasc
http://siepic.ubc.ca

lukasc-ubc, 2017-12-16:
The following method works:

https://github.com/lukasc-ubc/SiEPIC-Tools/commit/6dd23ee5d9185c2811d762df2c1b06a9d65062a2

bpoulse, 2017-12-16:
Added shortcuts to config-keybindings, solves issue.

lukasc-ubc, 2017-12-17:
This solution for some reason doesn't work on 0.24 (OSX, Python2).  Menus don't have the hotkeys.  I recall struggling with this as well, and concluded that the way to do it was using the pymacros and the config GUI.

bpoulse, 2017-12-17:
Alright, no sense wasting more time. I'll change it back to pymacros.

lukasc-ubc, 2017-12-17:
weird.  In settings the key bindings are correct, yet the menus are missing them.But when I push OK (with making any changes), then the menu gets updated correctly.  Perhaps there just needs to be a refresh command issued?

bpoulse, 2017-12-17:
Yeah, its something to do with the referencing we cant control very well. I've trying a ton of different things to try and get actions working but they seem very finicky. I am making pymacros for the menu items now

lukasc-ubc, 2017-12-17:
Ok, thanks!
I’ve found a few other bugs that I’ve fixed, for the path-waveguide functions.  
working on component snapping now.  I notice that your naming suggests that the action is different from what I originally had.

Previous “Shift-O”.  Select object you wish to move, then hover mouse to where you want it to go…

Yours: simpler - go to the nearest pin.

You think the simpler one is better / sufficient?  Did you find problems with the way I did it before?

thanks

bpoulse, 2017-12-17:
I didn't intend for it to be functionally different. I liked the way it was previously, had no issues. I havent done anything for component snapping, the function is defined in scripts.py but not implemented. The other snapping is implemented in extend.py for Paths, which just finds the nearest pin available (with correct rotation).

lukasc-ubc, 2017-12-17:
good.  I'll implement the same thing then...

bpoulse, 2017-12-17:
Changed to pymacros. If you don't mind verifying on your version then we can close.

lukasc-ubc, 2017-12-17:
Windows 7:

 - W, Shift-W, etc are working.
 - missing the "SiEPIC initialize" menu at the beginning.
 - you introduced a new error where the Waveguide (W) function doesn't work when trying to connect to a component.  (maybe I fixed something which you over-wrote in this commit?)

[image]

[image]

[image]

lukasc-ubc, 2017-12-17:
On OSX, version 0.24.10

 - SiEPIC Init menu is there...
 - same issue as Windows (W)
 - Shift-W has yet another error

[image]

[image]

[image]

lukasc-ubc, 2017-12-17:
I think the issue is with the following that you added in your last commit:

[image]

bpoulse, 2017-12-17:
Ah. I accidentally pushed untested changes to the map with it. I will fix it.

bpoulse, 2017-12-19:
@lukasc-ubc with removing that code and the other fixes you did is this closed?

closed 2017-12-20 by lukasc-ubc