SiEPIC/SiEPIC-Tools #31: Waveguide changes

bpoulse, 2017-12-19:
- Change LayerInfo list to actual names
- Change offset/slot to subtraction
- Guiding shape fix

lukasc-ubc, 2017-12-22:
I fixed 
 - Waveguide to path should read path width from PCell

the others to do.

lukasc-ubc, 2017-12-23:
Guiding shape. Looks like you've improved it, but there is a problem (perhaps not important).  When you instantiate a Waveguide PCell, then Partial select stretch it, the PCell doesn't update.

Interestingly, the one created by "W" works fine.

This is on 0.24.10.  Might be different on 0.25.

bpoulse, 2017-12-23:
Interesting. I'll investigate.

lukasc-ubc, 2017-12-23:
it's funny, each time we fix something, something else breaks (particularly with the new features in 0.25, which aren't there in 0.24).  hopefully my fixes of your fixes didn't add more problems.

nedanr, 2018-02-07:
[image]

mustafacc, 2018-02-08:
Hi nedanr, This issue is SiEPIC_IME PDK specific, it has been fixed on the master repository but not yet released on kLayout packages.

lukasc-ubc, 2018-02-08:
I just released the latest versions of SiEPIC-Tools, SiEPIC_IME_Library, and SiEPIC_EBeam_PDK for the Package Manager, which fixes this issue.

commit 0e8ad1a3 (2017-12-23): more fixes #31
closed 2018-02-08 by lukasc-ubc