SiEPIC/SiEPIC-Tools #32: arc in SiEPIC.utils produces angled ends

mustafacc, 2017-12-21:
https://github.com/lukasc-ubc/SiEPIC-Tools/blob/master/klayout_dot_config/python/SiEPIC/utils.py#L213

When used in the [following PCell ](https://github.com/lukasc-ubc/SiEPIC_EBeam_PDK/blob/c7e0c9919d2baa1f7f573194fe9e52df27e5608d/klayout_dot_config/tech/EBeam/pymacros/SiEPIC_EBeam-dev%20Library.lym#L1138)instead of layout_arc_wg_dbu , i'm seeing the following:

[image]

lukasc-ubc, 2017-12-21:
I fixed that. Which branch are you looking at?

Lukas Chrostowski
Professor, ECE
University of British Columbia
604-822-8507
http://www.mina.ubc.ca/lukasc
http://siepic.ubc.ca

lukasc-ubc, 2017-12-21:
Hi Mustafa,

I see it.  this is in dc_half_ring.  anywhere else?

lukasc-ubc, 2017-12-21:
Hi Mustafa,

This is indeed a problem with the arc & translation approach that Brett implemented.  

Rather than translating a path arc, I think we need to go back to a single polygon defined by two arcs. That way the corners are defined precisely.

lukasc-ubc, 2017-12-21:
Hi Mustafa,

It is now fixed.

We have two options now to draw the waveguide arcs:

    t = pya.Trans(pya.Trans.R0,Lc/2, -r)
    self.cell.shapes(LayerSiN).insert(arc_wg(r, w, 90-dc_angle, 90).transformed(t))

    self.cell.shapes(LayerSiN).insert(arc_wg_xy(Lc/2,-r, r, w, 90-dc_angle, 90))

Which do you prefer?  Keep both?

It is implemented in PCell: DirectionalCoupler_HalfRing_Arc
in EBeam v0.3.0

and arc_wg and arc_wg_xy are in Tools in the netlist-dev branch.

Also, I added "Merge all the waveguide shapes, to avoid any small gaps"
See the bottom of that PCell.

That library also needs to have the TECHNOLOGY changed, like we did elsewhere.

closed 2017-12-21 by lukasc-ubc