SiEPIC/SiEPIC-Tools #38: Path produced by arc_bezier creates angled ends

mustafacc, 2017-12-26:
https://github.com/lukasc-ubc/SiEPIC-Tools/blob/4de874bb38e922cceaa64f1e73681d721c140ce9/klayout_dot_config/python/SiEPIC/utils.py#L393

When used in:

https://github.com/lukasc-ubc/SiEPIC_EBeam_PDK/blob/ebae270a6fed1da10a7fbf8ba8879841d5aad467/klayout_dot_config/tech/EBeam/pymacros/SiEPIC_EBeam-dev%20Library.lym#L2503

lukasc-ubc, 2017-12-26:
Your implementation of the bezier is quite different from Brett's.  Can you compare them?  He does a "grow" operation which messes up the end points.  Yours does two separate arcs for the inside/outside; I think that approach is safer.

Perhaps let's merge your two codes together to have only one bezier implementation?

thanks

lukasc-ubc, 2018-01-05:
Brett fixed it.

closed 2018-01-14 by lukasc-ubc