SiEPIC/SiEPIC-Tools #59: Calibre DRC integration

lukasc-ubc, 2018-01-24:


lukasc-ubc, 2018-01-25:
Passwordless login instructions for OSX:
* https://help.dreamhost.com/hc/en-us/articles/216499537-How-to-configure-passwordless-login-in-Mac-OS-X-and-Linux

lukasc-ubc, 2018-01-25:
Brett's improvements: https://github.com/lukasc-ubc/SiEPIC-Tools/commit/c9316f9d34c771f1d70636df97b213adf1101836

lukasc-ubc, 2018-01-25:
The rule filename shouldn’t be hard-coded, but rather should be in the XML… (CMC_SiEPIC_IMESP.drc.cal)… to do.

lukasc-ubc, 2018-01-25:
an easier way to copy your key to remote server:

> ssh-copy-id -i ~/.ssh/id_rsa.pub username@server.dreamhost.com

Does this work on windows?

bpoulse, 2018-01-25:
There might be a command like it in some library of Cygwin, but I haven't found it. Since the other way worked on Windows I used that for the instructions, but it's not limited to it on the other systems.

lukasc-ubc, 2018-01-25:
It is implemented and working on both OSX and Windows.

Using ~/.ssh/config for the server, and CALIBRE.xml for the server commands/paths/PDK/rule files...