SiEPIC/SiEPIC-Tools #60: KLayout integration installation issue

GuanhuiWang, 2018-01-30:
Hi Lukas,

Following is a problem reported by a customer when try to install the integration:

"I first uninstalled the Klayout then installed it again and follow the way as told in GitHub

lukasc-ubc/SiEPIC-Tools

SiEPIC-Tools - Package for KLayout to add integrated optics / silicon photonics functionality (waveguides, netlist extraction, circuit simulations, etc)
I went to tools --> Manage Packages, --> installed the packages siepic_tools_0.3.18 and also siepic_ebeam_pdk 0.3.5

I have already installed Python and Numpy library on my computer.

Do I need to install the windows_python_package_for_kKlapout also ??
I run all the command line prompt which starts with mlink

I get this error when I first try to run SiEPIC 0.3.16 --> SImulation --> Setup Lumerical Interconnect Compact Model Library

[image]

I have very limited knowledge on this. Please help to comment on this problem. "

Thank you,
Guanhui

lukasc-ubc, 2018-01-30:
Yes, windows_python_package_for_KLayout is required for windows users.  It isn’t easy to get KLayout to use separately a installed Python installation (it would need to be exactly the same version as the one which the software was compiled with).

The mklink commands are only for developers.  For end users, the only requirement is KLayout 0.25, and SiEPIC-tools 0.3.22, and the windows packages.

GuanhuiWang, 2018-02-02:
Thank you Lukas. I will let the customer know.