SiEPIC/SiEPIC-Tools #62: Lost Library

MohamedElRayany, 2018-02-07:
I followed the installation instructions exactly as explained by this [wiki](https://github.com/lukasc-ubc/SiEPIC_EBeam_PDK/wiki/Installation-instructions)  and I am using INTERCONNECT 7.5 running on Windows Server 2012.
I still have a problem when I am working with the SiEPIC EBeam PDK. I cannot find all the components of the library when I add new istance.

[image]

How can I add the other components of the library such as the bdc, gc, waveguide straight,...etc

MohamedElRayany, 2018-02-07:
To solve this issue, one should select the technology to be EBeam after starting a new layout.
[image]

MohamedElRayany, 2018-02-07:
How could I add a **straight waveguide** cell from the EBeam Library?

lukasc-ubc, 2018-02-08:
Draw a Path (2 points). Then SiEPIC | Waveguides | Path to Waveguide.

MohamedElRayany, 2018-02-08:
Hi @lukasc-ubc 

I already tried this method. The created pins of the waveguide are not connected to the pins of the other cell. Here you can see a picture that shows the resulted errors after verifying the layout.
[image]

lukasc-ubc, 2018-02-08:
What is the cell on the bottom right?  For simulations and verification to work, each component needs the DevRec and PinRec layers. The one on the bottom right doesn’t seem to have this.

MohamedElRayany, 2018-02-08:
It is a **Waveguide** cell that can be found in **EBeam - Components & models [Technology EBeam]** library. I gave this method another trial to connect two different components (ebeam_gc_te1550 and ebeam_bdc_te1550). It did not give me connectivity error this time.

The old library **SiEPIC - EBeam PCells** contained many of the useful cells that do not exit anymore in the new versions of the EBeam libraries. Are there any reason why I can't use these cells? 
[image]

MohamedElRayany, 2018-02-08:
Hi Mustafa,

I tried to search for these particular cells, but unfortunately I can't find them under that mentioned library. 

[image]
PS: I am using the latest version

MohamedElRayany, 2018-02-08:
After doing an update to the library, I can see now the added **Waveguide_Straight** cell.

tatiana-pa, 2018-04-20:
Hi everyone,
I just installed the latest KLayout 0.25.2 and downloaded all SiEPIC packages using Tools->Manage Packages. When I add a new instance under EBeam technology I cannot find "spiral" component in the library (which I used in the old version of PDK for strip spiral waveguide of 500nm width). There are only several options with Spiral_BraggGrating. I am a Windows user, and currently use EBeam PDK 0.3.11.

[image]

How I can add a basic spiral component of the library? Thanks in advance.

StephenHLin, 2018-04-21:
Hi Tatiana,

The original spiral pcell seems to have been disabled due to the recent re-organization of our functions. I've written a new one for you and created a pull request for it.

The new pcell can be found under: Ebeam-dev -> SpiralWaveguide

StephenHLin, 2018-04-22:
I can confirm the SpiralWaveguide exists on my end:
Are you using the packaged releases (downloaded through klayout>tools>manage packages)? Or are you cloning it from github?

You're going to have to clone a version from github and install it manually if you want the SpiralWaveguide right away as the Klayout Packages release are usually only updated when large stable versions are approved.

tatiana-pa, 2018-04-22:
Yes, you are right. I was thinking the same but you are even faster in replying :) After manual updating everything works. Many thanks!

semihckm, 2018-07-02:
Nothing that I'm doing is working for 3 days:/ 
I'm so desperate and willing to pay

lukasc-ubc, 2018-07-03:
Can you please provide more details?  What are you trying to do. 

Lukas Chrostowski
Professor, ECE
University of British Columbia
604-822-8507
http://www.mina.ubc.ca/lukasc
http://siepic.ubc.ca

alexkarevsf, 2018-07-03:
Hi,

I am just trying to generate a gds file using Canopy and open it on
KLayout. But I can't make anything work. All the gds files are empty. I
found your update for 'Stream not found' error, and downloaded that but
still it isn't working. I tried to do all the fix I could find but I'm very
bad at this.

alexkarevsf, 2018-07-03:
For example, the example code below runs and generates an empty gds file.
Import lines do not generate any errors but when I add the write command I
see a hundred lines of warnings starting with  ValueErrorTraceback (most
recent call last) .

-----------

from ipkiss.all import *
import sys
from math import *

class ElementsExample(Structure):

     def define_elements(self, elems):
         for i in range (5, 50, 5):
                #circular arc
              elems += ArcPath(layer = Layer(0), center = (0,0),
                           radius  = i, start_angle = 0,
                           end_angle = 120, line_width = 1.0)
                #elliptical_arc
              elems += EllipseArcPath(layer = Layer(0), center = (120,0),
                           box_size = (2 * i, i) , start_angle = 45,
                           end_angle = 270, line_width = 1.0)
         return elems

layout = ElementsExample(name = "layout")
my_lib = Library(name = "ELEMENTS", unit = 1E-6, grid = 5E-9)
my_lib += layout
o= FileOutputGdsii("example_elements.gds")
o.write(my_lib)

lukasc-ubc, 2018-07-04:
This doesn’t look like a SiEPIC tools issue, but rather a problem with Canopy or ipkiss. I don’t even understand why you are also installing SiEPIC tools. Just go with a clean KLayout installation since you are only viewing the gds. 

Lukas Chrostowski
Professor, ECE
University of British Columbia
604-822-8507
http://www.mina.ubc.ca/lukasc
http://siepic.ubc.ca