SiEPIC/SiEPIC-Tools #69: Problem installing siepic addon

gabrilator91, 2018-02-22:
Hi, 

I am part of the EdX course, and I was trying to add the siepic pdk extension into klayout. 

I'm getting the following error
"Stream has unknown format", after being asked If I allowed the macros to be executed. 

After that, I can see the SiEPIC addon, but when I try to add the elements to my project, the SiEPIC library is empty. 

Could someone please give me a hand :) ? 
Thanks!
Gabriel

mustafacc, 2018-02-26:
Hi Gabriel,

Can you please ensure you are following those steps:

http://recordit.co/fK7AsDud0v

Make sure you click on the "T" icon above EBeam

lukasc-ubc, 2018-02-26:
https://github.com/lukasc-ubc/SiEPIC-Tools/wiki/Common-problems

lukasc-ubc, 2018-02-26:
Also please install 0.25.1 (the latest)

gabrilator91, 2018-02-26:
Thank you! It's fine now 😊

closed 2018-02-26 by mustafacc