SiEPIC/SiEPIC-Tools #83: single bus ring resonator

nedanr, 2018-03-21:
Hi,
As there is no single bus ring resonator (filter) in SiEPIC _IME library, I was wondering to know whether I can convert the double bus to the single one or vary the width of only one of the buses. I would be grateful if help me with this issue.
Thank you in advance.

mustafacc, 2018-03-21:
Hi Nedan,

If I'm understanding correctly, are you looking for a component similar to that of "ebeam_dc_halfring_straight" in the EBeam PDK (picture attached), except where you can vary the widths of the ring and the bus independently?

[image]

If you wish to design an all-pass filter (with no drop port), you can connect two 90 degree bends to that "ebeam_dc_halfring_straight" component to close the ring's loop.

Mustafa

nedanr, 2018-03-21:
Hi Mustafa,
Thank you for your reply. Actually, my device is a add-drop resonator with
a heater for tuning. I want to design two series ring resonators which are
connected on upper bus with a taper. So, the width of the upper bus
waveguide of the second ring is larger than the the first one, however the
other parameters such as ring width and the lower bus width are the same
for both.

lukasc-ubc, 2018-06-26:
Use one ebeam_dc_halfring_straight and connect a waveguide to the two ports to form a ring.

closed 2018-06-26 by lukasc-ubc

srinivas5290, 2021-02-16:
Hi,
 Is it possible to design CROW filter using "ebeam_dc_halfring_straight" or any alternative solution, because i am struggling with this for the past 2 weeks

mustafacc, 2021-02-16:
Hi @srinivas5290 , see the attached example using the pcells in the EBeam library. You will need to engineer the coupling gaps and rings radii to your CROW specs, which will also likely need a tuning element.
[EBeam_CROW.zip](https://github.com/SiEPIC/SiEPIC-Tools/files/5990546/EBeam_CROW.zip)