SiEPIC/SiEPIC-Tools #94: KLayout crash when going for SiEPIC_IME or SiEPIC_AMF tech to SiEPIC_EBeam or SiEPIC_GSiP

mustafacc, 2018-08-25:
when switching the tech from the toolbar, klayout crashes when going for SiEPIC_IME or SiEPIC_AMF tech to SiEPIC_EBeam or SiEPIC_GSiP as seen below:

[image]

lukasc-ubc, 2018-08-27:
I am observing the same issue on MacOS.  Will try to create a test file and submit to KLayout github.

lukasc-ubc, 2018-08-27:
@mustafacc 

can you please check with the latest version 0.25.4?  Strangely the problem doesn't exist between GSiP and EBeam, but it does from [AMF, SiEPIC_IME] and EBeam.  Any idea what could be causing this?

lukasc-ubc, 2018-08-27:
work-around: File | Reader Options - enable this window.

Then when you open a layout, it asks which technology the file is. this way you don't need to change technology. It seems to go to (default) at least, rather than some other random previous technology.

lukasc-ubc, 2018-08-27:
@mustafacc 
another work-around: when a layout is open, File | Layout Properties: change Technology, Ok.  No crash this way.

mustafacc, 2018-08-27:
@lukasc-ubc 
Bug is still there in 0.24.4 Win10 x64.

I tried to checkout SiEPIC Tools and all the PDKs to older versions and somehow the problem still persists. I'll keep trying to see what commit is causing this.

mustafacc, 2018-09-09:
issue at:
https://github.com/lukasc-ubc/SiEPIC_IME_Library/commit/c3c19c29cdbd950f769f52f5898d43efca9e82ce

mustafacc, 2018-09-09:
fixed issue on IME PDK and it appears to have fixed the issue with all techs
https://github.com/lukasc-ubc/SiEPIC_IME_Library/commit/fe324597b6bca803d438438cff41a11f4383546c

closed 2018-09-09 by mustafacc