SiEPIC/SiEPIC-Tools #95: Refactor Pin or Port

thomaslima, 2018-10-16:
Pin and Port should have the same functionality, and should be represented in python classes in the PDK. We should think really hard about what information a Port should contain so that we don't need to refactor it ever again.

Right now, the types of the ports are deduced from their name. And they only take one geometrical parameter: width. So clearly it does not accommodate special needs like non-standard optical waveguides, RF-trace ports...

This is a hard task because ports are useful to SiEPIC-Tools, but its layout parameters is PDK dependent.

Suggestion:
- [ ] Discuss in person or over an issue.
- [ ] Create a wiki with all important properties of a port.
- [ ] Refactor Pin or Port so they are the same.

lukasc-ubc, 2019-02-05:
It would be nice if Pins/Ports were a native KLayout object.

thomaslima, 2019-02-05:
Have we decided on what is a Pin yet? By native klayout object, do you mean something that can be saved and loaded from gds objects etc?

lukasc-ubc, 2019-02-06:
Hi Thomas

It would be tricky to make it GDS compatible but I am sure it can be done like PCells were implemented. 

My main concern is that we should coordinate this activity with Matthias. He may already be working on this or have it on his roadmap. And it would make sense if it was part of the KLayout software. 

Lukas Chrostowski
Professor, ECE
University of British Columbia
604-822-8507
http://www.mina.ubc.ca/lukasc
http://siepic.ubc.ca