SiEPIC/SiEPIC-Tools #103: Catalina and/or klayout working

atait, 2019-11-02:
Hi Lukas,

Starting a new thread here off of https://github.com/KLayout/klayout/issues/381

I am getting the same GSI warnings as you are. Is this affecting the workshop? If so, I could spend some time on it tomorrow.

The warnings do not seem to affect my stuff which is python 3 and non-GUI. The GSI was originally Ruby then made "general" with the addition of python. My image didn't contain ruby, so that is probably why it is warning. It's possible, unfortunately, that klayout GUI or GSI depends on ruby pieces and parts.

When I was trying to use SiEPIC sans GUI, I came up with an approach in lygadgets called `patch_environment`. SiEPIC code runs GUI functions on import, so, if you want to use it within pure python, you have to trick it into thinking there's a GUI. If the students are having trouble opening a window and loading a layout, then klayout should be modified, otherwise I think it would be easier to extend the environment patch in lygadgets.

-Alex

lukasc-ubc, 2019-11-02:
> Starting a new thread here off of KLayout/klayout#381 <https://github.com/KLayout/klayout/issues/381>

Hi Alex,

Workshop was fine as I told people to install 0.25.9.  0.26 is still too experimental (and not even on the downloads page yet for Mac OSX!)

Thomas and I discussed extensively and agreed to move to Python 3 only for SiEPIC-tools and PDKs, and to have these work both within KLayout as well as in an outside python.  

Lots of work ahead!  But it will be a good investment for the large systems we will be making.

Lukas

timoteoman3, 2020-02-12:
If there is no Catalina for 0.25.9, is it possible to supply a command to suppress these warnings? I also am having no affect on my functionality with Python 3.7 and no GUI. I'm just curious if anyone had a workaround since I've tried redirecting stdout, stderr and catching warnings but I could have done those wrong as well.

lukasc-ubc, 2020-02-13:
Not sure what you mean…  Can you explain more?  
Last night, I succeeded in getting KLayout 0.26.3 + SiEPIC-Tools working on the Mac (both Mojave and Catalina).  I pushed the changes last night.

timoteoman3, 2020-02-13:
I think I found a workaround solution for now. Basically, the problem I was getting was the same GSI warnings when running Klayout from the command line from my python script. 

Ex. 
```python
# test.py
os.system('path/to/klayout_executable -zz path/to/klayout_macro')
```

klayout_executable: the path to the klayout executable on my Mac
-zz: open in GUI-less mode
klayout_macro: the path to my klayout_macro.py file

I wanted to suppress the GSI warnings when calling Klayout which I can do by editing the Klayout call to this
```python
# test.py
os.system('path/to/klayout_executable -zz path/to/klayout_macro &>/dev/null')
```

The `&>/dev/null` removes any warnings that occur in the klayout process.