SiEPIC/SiEPIC-Tools #108: One technology can only support one Lumerical lib

JinWallner, 2020-02-25:
Hi, 

Currently the Component. has_model() function hard codes the component path as design kits::technology_name::component name.  In this case, we can only use one lib in interconnect design kits/technology_name. But for each component, we have lib specified in the DevRec layer. Can we change the following line in has.model()
return ("design kits::" + TECHNOLOGY['technology_name'].lower() + "::" + self.component.lower()) in INTC_ELEMENTS

into 
return (self.library.lower().replace('/','::') + "::" + self.component.lower()) in INTC_ELEMENTS
This way, we are not limited with one cml with fixed name.

Thanks,

Jin

lukasc-ubc, 2020-02-25:
Hi Jin,

We actually already use multiple CML libraries, e.g., AMF-PDK and AMF-Library, so I assumed that this was already possible.  But perhaps all our designs used one.  Now that I look at your comment, I wonder why we didn’t run into this problem before.  Yes, why don’t you make a git Pull request?

thanks

JinWallner, 2020-02-26:
I submitted a pull request.

Thanks,

Jin

closed 2020-11-07 by lukasc-ubc