SiEPIC/SiEPIC-Tools #111: Support multiple WG models for path to WG

JinWallner, 2020-02-27:
Hi,

Can we add additional information in WAVEGUIDES.xml, such as CML and model for different WGs? When we execute path to WG, we can pick up those information and print the corresponding CML and model information in the WG?

Thanks,

Jin

lukasc-ubc, 2020-03-08:
Dear Jin,

Yes, we can do what you describe.  You’ll need to make your own waveguide model.  And we will need to add a field to WAVEGUIDES.XML to pass the model parameter, perhaps via “Component”.  We will need to edit the Waveguide PCell to check for this parameter and pass it on as a Spice parameter.

Let me know if you make progress on this, and do a pull request when you get it working!

regards
Lukas

JinWallner, 2020-03-09:
Hi Lukas,

I created a pull request to enable 2 new parameters CLM and model in the WAVEGUIDES.xml . I tested on my technology with updated WAVEGUIDES.XML and Waveguide PCell. It should be comparable with the technologies without using the 2 parameters.

Thanks,

Jin

closed 2020-11-07 by lukasc-ubc