SiEPIC/SiEPIC-Tools #122: Waveguide turtle routing

lukasc-ubc, 2020-06-29:
@SiEPIC/siepic-tools @mustafacc @SiEPIC/ubc-queens, @kwadwo00,

I implemented an easier way of creating waveguides for procedural scripted layout.  I would love your feedback.  Here is an example script:

```

# Copyright 2020, Lukas Chrostowski
# 2020/06/29, 00:47 Pacific/Cascadia time

# Scripted layout generator in KLayout
layoutname = 'test'

# Uses technology:
technology_name = 'EBeam'

# Create a layout
import pya
from pya import Trans, CellInstArray
cv = pya.Application.instance().main_window().create_layout(technology_name,1)
topcell = cv.layout().create_cell(layoutname)
cv.cell=topcell
print(cv.cell_name)

# Import functions from SiEPIC-Tools, and get technology details
import os
from SiEPIC.utils import get_layout_variables
TECHNOLOGY, lv, ly, cell = get_layout_variables()
dbu = ly.dbu
from SiEPIC.extend import to_itype
from SiEPIC.scripts import path_to_waveguide

# Layout generation function
def layout_generator():
    global cell, ly

    # Import Library cells (fixed)
    cell_ebeam_crossing4 = ly.create_cell('ebeam_crossing4', 'EBeam').cell_index()

    # Import Library cells (PCells, parameterized)

    # Place instances of cells
    inst_ebeam_crossing4_0 = cell.insert(CellInstArray(cell_ebeam_crossing4, 
            Trans.from_s('r270 0,0')))
    inst_ebeam_crossing4_1 = cell.insert(CellInstArray(cell_ebeam_crossing4, 
            Trans.from_s('r270 30000,20000')))
    inst_ebeam_crossing4_4 = cell.insert(CellInstArray(cell_ebeam_crossing4, 
            Trans.from_s('r270 0,-50000')))
    inst_ebeam_crossing4_5 = cell.insert(CellInstArray(cell_ebeam_crossing4, 
            Trans.from_s('r270 30000,-20000')))

    # Waveguides
    from SiEPIC.scripts import connect_pins_with_waveguide
    waveguide_type='Strip TE 1550 nm, w=500 nm'

    # Connect two components with a waveguide
    connect_pins_with_waveguide(inst_ebeam_crossing4_0, 'opt3', inst_ebeam_crossing4_1, 'opt3', turtle_A=[5.0, -90, 10.0, -90], turtle_B=[10.0, 90, 10.0, -90], waveguide_type=waveguide_type, verbose=True)

    # Connect two components with a waveguide
    connect_pins_with_waveguide(inst_ebeam_crossing4_0, 'opt2', inst_ebeam_crossing4_1, 'opt4', waveguide_type=waveguide_type)

    # Connect two components with a waveguide
    connect_pins_with_waveguide(inst_ebeam_crossing4_4, 'opt2', inst_ebeam_crossing4_5, 'opt3', waveguide_type=waveguide_type)

    # Connect two components with a waveguide
    connect_pins_with_waveguide(inst_ebeam_crossing4_4, 'opt4', inst_ebeam_crossing4_5, 'opt4', waveguide_type=waveguide_type)

    # Connect two components with a waveguide
    connect_pins_with_waveguide(inst_ebeam_crossing4_4, 'opt3', inst_ebeam_crossing4_5, 'opt2', turtle_A=[5.0, 90, 30.0, 90, 54.6, 90, 80.0, 90], turtle_B=[5.0, 90, 10.0, 90, 10.0, -90, 10.0, -90], waveguide_type=waveguide_type, verbose=True)

# Run the layout generator script
layout_generator()
```

And it generates the following layout:

[image]

lukasc-ubc, 2020-08-29:
@mustafacc have you tried this?

mustafacc, 2020-08-29:
@lukasc-ubc  Yes I did, works very nicely

closed 2020-11-07 by lukasc-ubc