SiEPIC/SiEPIC-Tools #129: Interconnect on HighSeirra

vanderpol2, 2020-08-06:
I am working through the lumerical INTERCONNECT process on OSX 10.13 
I have the following screen shots, 
i did this;
[image]

I am getting this error message;
[image]

I doubleclicked the file:
[image]
[image]

Lou

lukasc-ubc, 2020-08-06:
what folder you do you have lumerical installed in?  KLayout SiEPIC-Tools should automatically detect it.  I have never seen anyone go through what you are describing.  The wiki page was all that most people need…

vanderpol2, 2020-08-06:
Morning Lukas,
Thank you for the note,  I am going through the Wiki on compact
library installation for Ebeam now.
Hopefully that resolves the issue of compact library installation.
==
Initially I just downloaded the zip file and just opened that in my user
root.
---
Currently:
SiEPIC-Tools finishes installation successfully ( w warnings on python).
Step1 and Step 2  from the WIKI are fine except py2 warning.
Step 3: compact library.
not correct. AS noted working through wiki ,To answer your question
currently:
[image: Screen Shot 2020-08-05 at 8.43.36 AM.png]
 the directories are as shown.
[image: Screen Shot 2020-08-06 at 7.03.06 AM.png]
[image: Screen Shot 2020-08-06 at 7.03.06 AM.png]
=====

Thanks again for the advice. I will work through EBEAM WIKI and get
the compact library ebeam working.
=====
Then step back and go through some simple structure explicit "lego"
assembly processes.
First work through detail of layers, materials boundaries, .netlists. etc
on the examples provided.
Then build "LEGO" style waveguide, splitters,( a heater,) grating couplers.
Create a notebook for all the details.
build out some example test structure from there.
Then step through the  features in SiEPIC ( measure waveguide, etc,lots of
great stuff in there)
Eventually working back up  to build sets of tightly spaced mzi etc.
=====

Best
Lou

vanderpol2, 2020-08-06:
Your WIKI is awesome.
Thanks
Lou

lukasc-ubc, 2020-08-29:
@vanderpol2 

Sorry I was away for a while.  Did you manage to solve things?

vanderpol2, 2020-08-29:
Hi Lukas, 
I am pretty good, frying pan to fire at the moment good hold status on kaylout. Alan Mickelson asked me to take AIM SiP circle 1 Stefan Preble class, I said ok and it has been interesting and challenging, at modular construction  it took the RSoft way as I built devices with Dominic  Gallagher’s FIMMWAVE and Fimmprop tools in the distant past and on Si slot mode electro-optic polymer devices 
Your class I think starts Sept15 and depending on the state of my circuit I I was hoping to go through your course for the kaylout and your designs are very interesting, tbd learning much more then expected and there is a bit of a fire hose of info video after video awesome but a buggy 
Look forward to our course audit at a minimum
Best Lou