SiEPIC/SiEPIC-Tools #136: Error: 'Waveguide' in 'SiEPIC General' library is not available.

tmcnitt, 2020-12-15:
Error occurs when trying to use Waveguides -> Path to Waveguide

Full error
`'Waveguide' in 'SiEPIC General' library is not available. Check that the library was loaded successfully.
  /Users/username/.klayout/salt/siepic_tools/python/SiEPIC/scripts.py:524
  /Users/username/.klayout/salt/siepic_tools/pymacros/Keybindings/Waveguides/Path to Waveguide - Hotkey.lym:5
`

Klayout version 0.26.9
SiEPIC version 0.3.71 Py 2

mustafacc, 2020-12-15:
Hi Troy, Which PDK are you experiencing this issue with?

achuify, 2020-12-15:
I am part of the same research group as tmcnitt. It’s the AIM Photonics PDK 4.0c.

mustafacc, 2020-12-15:
Unfortunately this is a SiEPIC PDK-related issue and cannot be fixed from SiEPIC-Tools. Since the SiEPIC AIM PDK is not hosted on GitHub (i believe whatever version you might have is hosted locally), I have no way of debugging the version on your end. The version of the SiEPIC AIM PDK I have on my end does not have that issue.

Perhaps you can get your project's supervisor to contact us at info@siepic.com if your group cannot figure out how to look into the Python backend of the PDK to fix this. We're limited on what we can share and discuss in this public forum due to confidentially agreements.

lukasc-ubc, 2020-12-17:
Can you please check that on the bottom left of the screen, you see a “T” with the AIM Photonics PDK selected?

tmcnitt, 2020-12-17:
Hi, thanks for your help on this. We downgraded to version v0.3.69 and the issue disappeared.

closed 2020-12-17 by tmcnitt

lukasc-ubc, 2020-12-17:
Thanks Troy.  Good feedback.  I’m on version 0.3.72.  You also mention Python 2.  There are so many flavours of KLayout.  I’ll see if I can reproduce this on my end and make sure it works in the later versions.