SiEPIC/SiEPIC-Tools #137: get_technology_by_name function crashes when cml file doesn't have _v in the name

JinWallner, 2020-12-23:
For v0.3.70 and v0.3.71, the get_technology_by_name function expects the cml file with the naming convention name_vxxx.cml. If there are some cml files not following the naming convention, the get_technology_by_name function will crash. Can we have a default version in case the cml file name doesn't have _v?

Thanks,

Jin

lukasc-ubc, 2020-12-24:
Sure, do you have a suggestion?  We can also skip the CML file, and leave it as “None” for a default.

JinWallner, 2020-12-28:
Hi Lukas,

In the get_technolgy_by_name function, can we replace the following line at
2 locations

*cml_name, cml_version = file.split('_v', 1) *

with

* try:*

* cml_name, cml_version = file.split('_v', 1)*

* except:*

* cml_name=file[:-4]*

* cml_version="default.cml"*

It seems to fix my technology loading issue when there are cml files but
the cml file name doesn't have '_v' in it. I am not sure the default
version needs to be anything special for any downstream process. The code
already handles technology without any cml files.

Thanks,

Jin