SiEPIC/SiEPIC-Tools #147: Portions of libraries missing due to errors in installation macros (?)

karl-johnson, 2021-05-26:
I apologize if this is the wrong place to put this or if this has been answered before, but I have not seen this problem in any other of the issues on this repo (the closest is #62 ). When I attempt to install both "SiEPIC-Tools" (version 0.3.75) and "SiEPIC_EBeam_PDK" (version 0.3.33) through the KLayout package manager, errors appear in the installation window after I click "Yes" on the following dialog box:
[image]
For SiEPIC-Tools, I get these errors:

```
C:/Users/Kyle/KLayout/salt/siepic_tools/tech/GSiP/pymacros/GSiP Library.lym:64: [WinError 123] The filename, directory name, or volume label syntax is incorrect: "C:\\Users\\Kyle\\AppData\\Roaming\\KLayout\\'C:\\Users\\Kyle\\KLayout\\salt\\siepic_tools\\tech\\GSiP\\pymacros\\pcells_GSiP"
 C:/Users/Kyle/KLayout/salt/siepic_tools/tech/GSiP/pymacros/GSiP Library.lym:64 (class OSError)

```
For the PDK, I get these errors:

```
C:\Users\Kyle\KLayout\salt\siepic_tools\python\SiEPIC\utils\__init__.py:99: 'NoneType' object has no attribute 'technology'
  C:\Users\Kyle\KLayout\salt\siepic_tools\python\SiEPIC\utils\__init__.py:99
  C:\Users\Kyle\KLayout\salt\siepic_tools\python\SiEPIC\utils\__init__.py:228
C:\Users\Kyle\KLayout\salt\siepic_ebeam_pdk\tech\EBeam\pymacros\pcells_EBeam\contra_directional_coupler.py:28
  <string>:1
C:/Users/Kyle/KLayout/salt/siepic_ebeam_pdk/tech/EBeam/pymacros/SiEPIC_EBeam_Library.lym:166
C:/Users/Kyle/KLayout/salt/siepic_ebeam_pdk/tech/EBeam/pymacros/SiEPIC_EBeam_Library.lym:180 (class AttributeError)
C:\Users\Kyle\KLayout\salt\siepic_tools\python\SiEPIC\utils\__init__.py:99: 'NoneType' object has no attribute 'technology'
  C:\Users\Kyle\KLayout\salt\siepic_tools\python\SiEPIC\utils\__init__.py:99
  C:\Users\Kyle\KLayout\salt\siepic_tools\python\SiEPIC\utils\__init__.py:228
C:\Users\Kyle\KLayout\salt\siepic_ebeam_pdk\tech\EBeam\pymacros\pcells_EBeam_Beta\Bezier_Bend.py:21
  <string>:1
C:/Users/Kyle/KLayout/salt/siepic_ebeam_pdk/tech/EBeam/pymacros/SiEPIC_EBeam_Library_Beta.lym:187
C:/Users/Kyle/KLayout/salt/siepic_ebeam_pdk/tech/EBeam/pymacros/SiEPIC_EBeam_Library_Beta.lym:201 (class AttributeError)
```
Now, it appears to me that much of the package is still functional despite this, but the libraries (one of my main reasons for using the package) appear not to have been fully installed as a result of the errors (note - this is how the window looks even when selecting EBeam as the technology when creating a new layout, as mentioned by #62 ):

[image]

(another note - this is what I get after ~3-4 reinstallations of the package. Before the most recent install, the EBeam library was not present at all). 

Though I think I might be able to do what I need to work on using the cells that are there, I'm fairly certain there are a few cells (such as the cell containing a whole ring resonator and two bus waveguides) that are missing. Did I do something wrong during installation, or is this an issue with the macros run during the installation of the package?

lukasc-ubc, 2021-05-26:
Which KLayout version are you running?  

0.27 is not yet supported, and I get this same error in this version.

lukasc-ubc, 2021-05-26:
On Mac OSX, I have a similar error, which seems to be related to a python os realpath function:  https://github.com/KLayout/klayout/issues/810

karl-johnson, 2021-05-26:
I was running 0.27, as I installed it today. Sure enough, downgrading to 0.26.12 fixed all the issues. Thanks for the quick response!

karl-johnson, 2021-05-27:
I'm not re-opening this issue, as the package _appears_ functional for me right at this moment, but it seems like this could change at any instant - errors are magically appearing then fixing themselves and that scares me. For one, when I launched KLayout this afternoon, all the libraries I saw yesterday were gone. Checking File -> Log Viewer, I had these errors:
[image]. 

Clearly, these issues can be solved simply by adding "import sys" to these files. Indeed, upon doing so, I had access to the PDK libraries. Unfortunately, I was completely unable to place any cells from these libraries. Nothing happened when I tried to place an instance, and in the Log Viewer, I got these errors:
[image]
Nothing would place. In response, I tried uninstalling and reinstalling the package, as well as completely reinstalling the program. Neither worked. 

So, as I was downloading an even older version (0.26.10) of KLayout to see if _that_ one would work, I decided to also remote into my work computer and try on yet another version (0.26.9). Indeed, I got the same errors regarding sys and fixed those with "import sys", but this time, placing components worked perfectly. Returning to my home computer, _without changing my 0.26.12 install_, I was suddenly able to place components there too. The only change I made on my home computer was uninstalling and reinstalling the SiEPIC packages and manually doing the "import sys" modification again. I'm happy that the program is working right now, but I don't know how long this will last. Looking in the logs, I now see more new errors:

[image]

I'm guessing these will also prevent some other functionality in the program from working, but I haven't been able to do much actual layout yet, so I don't know what it's going to be this time.

closed 2021-05-27 by karl-johnson

lukasc-ubc, 2021-07-02:
@karl-johnson 
Are things working for you?  I just added "import sys" to the EBeam libraries.