SiEPIC/SiEPIC-Tools #148: Opt_in Label

flynnfd, 2021-06-22:
[image]

Hello,

I am attempting to simulate the performance of the Simple MZI structure on interconnect; however, K Layout reports that no components are connected to the opt_in label (see image). How do I connect components to the opt_in label?

lukasc-ubc, 2021-06-22:
create a text label, with “opt_in_TE_1550_xxxxx” on the Text layer, located at the tip of one of the grating couplers.
You can also see the examples in the GitHub repository for the SiEPIC EBeam PDK.

flynnfd, 2021-06-23:
Perhaps I am misunderstanding your instructions. I added a text label to the tip of my grating coupler as shown below.

[image]

After attempting the simulation I receive:

[image]

When is a good time to contact you? I can share my screen with you and we can resolve this issue in a timely fashion.

jaspreetj, 2021-06-23:
@flynnfd If you are using the laser element from SiEPIC General library, are you defining probing points/ outputs using the detector elements?

flynnfd, 2021-06-23:
I am using the laser element from the SiEPIC General Library. I am fairly new to this software so I most likely did not define probing points. Is there a time I can contact someone to share my screen in real-time?

jevillegasd, 2022-08-17:
This comes late, but it seems to me that the waveguides were just a polygon (I cannot see the DevRec boundary around them). If that's correct, then perhaps you just needed to make new waveguides using the Path 2 Waveguide command.

mustafacc, 2022-08-17:
Good catch. Thanks!