SiEPIC/SiEPIC-Tools #150: Suggest to add a second WAVEGUIDES.xml file for custom waveguide

JinWallner, 2021-07-06:
Hi,

Can we add a second WAVEGUIDES.xml file, e.g. WAVEGUIDES_custom.xml for custom waveguide configuration? Currently, custom waveguides and PDK waveguides share the same file. It needs to be remerged for every new PDK release.

Thanks,

Jin

mustafacc, 2021-07-06:
Good idea, perhaps instead of importing WAVEGUIDES.xml we should just import every xml file containing "WAVEGUIDE".

lukasc-ubc, 2021-07-08:
I have implemented this.  I loads WAVEGUIDES.xml and then all WAVEGUIDE_*.xml files.  It is on GitHub, but not released in Package Manager. Can you please check?

mustafacc, 2021-07-08:
Works okay for EBeam but there seems to be issues with AMF and AIM

AMF: generated waveguides missing pinrec layers + asterisk
[image]

AIM: waveguides don't generate properly
[image]

lukasc-ubc, 2021-07-08:
Hi Mustafa,
Thanks for checking.  I don’t see this problem on my end.  Let me investigate further.

lukasc-ubc, 2021-07-08:
I found an issue where a string representation of the bezier parameter caused the waveguide to not generate. Just fixed. Before updating, can you check if this problem only occurred for Bezier curve waveguides?  i.e., you shouldn't have had this issue with non-bezier waveguides (e.g., TM).