SiEPIC/SiEPIC-Tools #151: Issue with Setup Lumerical INTERCONNECT Compact Model Library with KLaout 0.27.2

JinWallner, 2021-07-06:
Hi,
For Klayout 0.27.2, when I click KLayout>SiEPIC > Simulation, Circuits > Setup Lumerical Interconnect compact Model Library, it gives an error message saying "Local Variable 'libraries' referenced before assignment". 

It seems that the **pya.Library.library_by_name** in Setup_Lumerical_KLayoutPythong_intergration in interconnect.py doesn't work with lib name alone anymore. It will require a technology name as well. This Klayout update leaves **libraries** undefined. A quick fix will be assigning an empty list to **libraries** in the except. 

Another way to fix it will be using the following to do the assignment:
**libraries = [n for n in pya.Library.library_names() if pya.Library.library_by_name(n,TECHNOLOGY['technology_name'] ) ]**
But this will not work with older Klayout and will require different ways to handle the old and knew klayout.

Thanks,

Jin

lukasc-ubc, 2021-07-07:
Hi Jin,

We can fix this.  

But probably the best would be for users to install Lumerical CMLs manually directly within INTERCONNECT.  This way they know where it is installed, and doesn't rely on KLayout doing the installation.

JinWallner, 2021-07-07:
Hi Lukas,

Install Lumerical CMLs manually should be ok. But the SiEPIC tool checks
whether the component has a Lumerical model or not. So we will still need a
way to keep track of the components' model.

Thanks,

Jin

lukasc-ubc, 2021-07-07:
Hi Jin,

I have removed the check for the compact model existence in https://github.com/SiEPIC/SiEPIC-Tools/releases/tag/v0.3.77, since it was often a source of confusion for designers.  

If this feature is useful, we could make it optional and configured as part of the PDK.

JinWallner, 2021-07-07:
Hi Lukas,

Thank you for the explanation. It makes sense now. I will update the PDK
instruction to have the users install the CML lib manually inside the
INTERCONNECT.

Thanks,

Jin