SiEPIC/SiEPIC-Tools #152: How to make .svg shown  correctly in function "Component FDTD simulation for S parameter,create Compact Model"

Zwlzzm, 2021-07-07:
Hi,

Convert the polygon in the layout to .svg seems to be achieved by the following code, but the svg file and port appearing in interconnect are mismatched. Why?

[image]

    TECHNOLOGY = get_technology()

    # get polygons from component
    polygons = component.get_polygons(include_pins=False)
    x, y = component.DevRec_polygon.bbox().center().x, component.DevRec_polygon.bbox().center().y
    width, height = component.DevRec_polygon.bbox().width(), component.DevRec_polygon.bbox().height()
    scale = max(width, height) / 0.64
    s1, s2 = (64, 64 * height / width) if width > height else (64 * width / height, 64)

    polygons_vertices = [[[round((vertex.x - x) * 100. / scale + s1 / 2, 2), round((y - vertex.y) * 100. / scale + s2 / 2, 2)]
                          for vertex in p.each_point()] for p in [p.to_simple_polygon() for p in polygons]]

    for i in range(0, len(polygons_vertices)):
        if verbose:
            print('polygon: %s' % polygons_vertices[i])
        p = dwg.add(dwg.polyline(polygons_vertices[i], fill=color, debug=False))  # stroke=color

    dwg.save()

    if os.path.exists(svg_filename):
      t+= 'seticon(component,"%s");\n' %(svg_filename)
    else:
      print(" SiEPIC.lumerical.fdtd.component... missing SVG icon: %s" % svg_filename)

I tried to modify the polygons_vertices, then .svg file will be correct for one pcell, but .svg file is still mismatched for annother pcell.

    polygons_vertices = [[[round((vertex.x - x)* 100. / scale + s1 / 4+0.75*s1/10, 2), round((y - vertex.y)  * 100./ scale + s2 / 4-s2/10, 2)]
                          for vertex in p.each_point()] for p in [p.to_simple_polygon() for p in polygons]]

Is there a way to make it correct for every pcell or most pcells?

Beside, i found .svg file doesn't show in this function "Component FDTD simulation for S parameter,create Compact Model" for the lastest version of siepic tools, are there some consideration for not including .svg file?

Could someone give me some of his opinion?
Thank you in advance!

lukasc-ubc, 2021-07-09:
The SVG generation hasn't changed in SiEPIC-Tools in a while, unless one of the underlying Python libraries has changed.  Looking at the above code, it looks like the polygons are being centred, so the origin the polygons should be the midpoint of the layout.
Perhaps Lumerical has changed the convention for the origin of the SVG.  Maybe try editing the SVG file, moving the object, and manually loading it into INTERCONNECT.  You could also check the Lumerical documentation or technical support to find out what the SVG origin requirements are, and we can modify the above code to fix it.

thank you

Zwlzzm, 2021-07-09:
@lukasc-ubc,
Thank you very much for your reply.  
Changing svg file manually is a way,but not perfect.
Actually I am not  familiar with the underlying python, I will ask question on lumerical support website to see whether they have any good suggestions.
Thank you again for your advice.

adophobr, 2022-02-24:
Hi @Zwlzzm,

The problem is the format of the second argument in 
svgwrite.Drawing(filename, size=(str(s1) **+ '%'**, str(s2) **+ '%'**), debug=False)
The *.svg file is saved with a size of 300x150 px. Using (without "%")
svgwrite.Drawing(filename, size=(str(s1), str(s2)), debug=False)
the *.svg file is saved with correct size (for example 64x30.27027 px for ebeam_y_1550 and 64 x 64 px for ebeam_crossing4).

closed 2022-07-26 by mustafacc