SiEPIC/SiEPIC-Tools #153: Automatic SBend

lukasc-ubc, 2021-07-09:
Automatically detect when an SBend is needed in the Waveguide.

jevillegasd, 2022-08-18:
For this, should it insert an SBend PCell on site (ask the user?)  or generate an SBend curve as part of the waveguide?

mustafacc, 2022-08-18:
I think either is okay. It can be implemented in a similar manner to compound waveguides in which a taper cell is automatically instantiated 
https://github.com/SiEPIC/SiEPIC_EBeam_PDK/blob/a45f05eab1f27c842352d5f0ab44987876b52a99/klayout_dot_config/tech/EBeam/WAVEGUIDES.xml#L232

What do you think?

jevillegasd, 2022-08-18:
SBends PCells are pretty generic provided the PDK has a Cell for it, so it should be possible to insert the cell as the compound does without the need to specifying it in the waveguides.xml.

But I think it might be easier to just use the geometry.bezier.parallel function to add the bend in place as a list of points to append to the ones that define the wg polygon.

I guess there still needs to be a minimum SBend radius to be calculated from the original bending radius?

jevillegasd, 2022-08-20:
With the changes I made, because I'm modifying layout_waveguide2(), opening an older file would insert SBends wherever they are found, as long as they were created after # 118 . This is probably an unwanted behavior, so to avoid it I could include a 'insert sbend' parameter in the waveguide PCell, and check whether that parameter exists or not before inserting the SBend. What do you think?

jevillegasd, 2023-01-17:
Hey @lukasc-ubc I think we can close this issue, addressed in  #179. **However**, the current version of the automatic SBends uses the  bezier_parallel function that is pretty slow. So maybe an uodate on either to make it more efficeint will help its usage be more smooth.

Waveguide definitions need only to include the follwoing parameter to allow for automatic sbends:
```
<sbends>y</sbends>
```

lukasc-ubc, 2024-02-14:
@jevillegasd I really like the automated SBends.  

In terms of performance, I added some lrcache functions late last year. Can you check if that improved the performance?  

For verification, indeed we need to remove the min-radius error for cases where the sbend is instantiated.

commit 493a72d6 (2025-09-26): "W" Waveguide GUI error check fix

#153