SiEPIC/SiEPIC-Tools #158: Siepic Tools not able to be found in salt

alexsludds, 2021-09-15:
Hi all,
I am trying to reinstall siepic tools in salt on klayout 0.27.3
However, it does not appear on the list of available packages.
If I try to install siepic_ebeam_pdk it says that it can't find siepic tools in the package index.

alexsludds, 2021-09-20:
Sorry. I'm going to be "that guy" and bump this just because I need Siepic tools working soon for a tapeout I am doing.

alexsludds, 2021-10-04:
Hi folks, is there just not people maintaining this software anymore? It's been incredibly helpful for my graduate research and saved me countless days of time. I would hate to see it become "abandonware"

dichium, 2021-12-06:
Dear Mr. @mustafacc 

Please help me in solving the problem of not being able to connect the component library in EBeam v0.3.37. I want to learn about MZI and want to simulate on KLayout for my graduate research.  I use windows 10 and have managed packages as I have seen on this forum. is there a step I missed? so my KLayout can't connect between respective device libraries. thank you please help.
[image]

lukasc-ubc, 2021-12-06:
Could you please use KLayout 0.27.5?  We stopped supporting 0.26.

closed 2022-07-26 by mustafacc