SiEPIC/SiEPIC-Tools #161: Cannot connect between any library component

dichium, 2021-12-01:
Please help me in solving the problem of not being able to connect the component library in EBeam v0.3.37. I want to learn about MZI and want to simulate on KLayout. I followed the tutorial from Mr. Subhash Arya (https://www.youtube.com/watch?v=ATHSvarYMOw&list=LL&index=11&t=1053s). I use windows 10 and have managed packages as I have seen on this forum. is there a step I missed? so my KLayout can't connect between respective device libraries. thank you please help.
[image]

aryasubh, 2021-12-01:
Thank you for watching the video. To connect a component with other components, please select a component ( Y-branch) which you want to connect by moving to a stationary component (Grating Coupler ) and then hover your mouse over the grating coupler and then press Shift+o. I hope this will work. Best wishes.

lukasc-ubc, 2021-12-11:
@dichium 
The issue might be because you are using KLayout 0.26.  We are not supporting 0.26 anymore.  I know that "technology" is a feature that was introduced in 0.27.  Can you please try upgrading?
thank you

closed 2022-07-26 by mustafacc