SiEPIC/SiEPIC-Tools #162: Error "Layout object  has no attribute  technology" when using "Path to Waveguide" function

cheng1214, 2021-12-05:
Hi All, 

After I updated the SiEPIC packages, now everytime I use "path to waeguide", it threw an error "Layout object  has no attribute  technology",

Details is as below:

'Layout' object has no attribute 'technology'
  C:\Users\Rui Cheng\KLayout\salt\siepic_tools\python\SiEPIC\extend.py:688
  C:\Users\Rui Cheng\KLayout\salt\siepic_tools\python\SiEPIC\scripts.py:515
  C:/Users/Rui Cheng/KLayout/salt/siepic_tools/pymacros/Keybindings/Waveguides/Path to Waveguide - Hotkey.lym:5

Could anyone help me fix this?
Thanking you!
Rui

closed 2021-12-05 by cheng1214

lukasc-ubc, 2021-12-05:
You closed the issue.  Is it still a problem?
thank you

cheng1214, 2021-12-06:
Hi Lukas，

Although not confirmed yet, I think the issue is because I was using KLayout 0.26.8, while "Technology" method is not available untill Klayout 0.27

According to Description of "Technology" method in Class Layout on （https://www.klayout.de/doc/code/class_Layout.html#method127)
>"This method has been introduced in version 0.27. Before that, the technology has been kept in the 'technology' meta data element."