SiEPIC/SiEPIC-Tools #163: KLayout cound not find SiEPIC EBeam library and Beta library when mannually instaililing SiEPIC PDK and Tools

cheng1214, 2021-12-06:
Hi All.

Due to the connection restrictions, I have to mannually install the SiEPIC PDK and Tools. The version is 0.27.5

What I did was:
1. create a new "salt" folder inside KLayout config folder
2. copy "Python_packages_for_KLayout" in "salt"
3. create two new folders in salt, called "siepic_ebeam_pdk", and "siepic_tools", and copy their corresponding "klayout_dot_config"  into them.

But after such installation,  KLayout cound not find SiEPIC EBeam library and Beta library (that is there is no such libraries in Instance).

Also, now when I ran SiEPIC_EBeam_library.py or SiEPIC_EBeam_library_Beta.py in Macro, it threw an error 
"[WinError 123] Syntax error in file name, folder name or volume label : "C:\\Users\\Rui Cheng\\AppData\\Roaming\\KLayout\\'C:\\Users\\Rui Cheng\\KLayout\\salt\\siepic_ebeam_pdk\\klayout_dot_config\\tech\\EBeam\\pymacros\\pcells_EBeam"

Could anyone help me fix the this problem?
Thanking you!

cheng1214, 2021-12-07:
I just found that in my KLayout environment,  __file__ returns 'path' , not path.

That is the reason why os.path.realpath(__file__)  return invalid path.

But I still don't know why this weird problem happens....

lukasc-ubc, 2021-12-07:
Hi Rui,

In the KLayout macro window, using the Python interpreter, can you please try this?

Turn on debugging mode, by clicking on the little bug (3rd from right icon).  If you get errors, you can debug them this way.

Open the SiEPIC_EBeam_Library.lym file and run it.  Then:

> import os
> os.path.realpath(__file__)

'/Users/lukasc/SiEPIC_EBeam_PDK/klayout_dot_config/tech/EBeam/pymacros/SiEPIC_EBeam_Library.lym’

Your error might be because of the space in your username, where Windows/KLayout/Python aren’t handling it correctly.

Good luck!
Lukas

cheng1214, 2021-12-08:
Hi Lukas,

Thanking you very much! I think you are right that the issue is due to the space of username.  I will try to fix it.

Best Regards,
Rui

lukasc-ubc, 2021-12-11:
@cheng1214 
let me know if that fixes it?
This might be a bug in KLayout rather than SiEPIC-Tools, so perhaps you could report it to github.com/klayout/klayout?

thanks

cheng1214, 2021-12-15:
@lukasc-ubc  Hi Lukas, 

This issue doesn't exist in another laptop of mine  where the username has no name psace

I have fixed it by removing " ' " and " ' " at both the end of the path string, i.e., "__file__[1:-1]", such that it becomes valid.

I have reported the issue in KLayout (https://github.com/KLayout/klayout/issues/946)

closed 2022-07-26 by mustafacc