SiEPIC/SiEPIC-Tools #170: 'Waveguide' is not available. Check that the library was loaded successfully.

hasanhoji, 2022-06-25:
I am using Klayout 0.27.10, SepicTools 0.3.91, Siepic_ebeam_pdk 0.3.42, NanoSOI_PDK_v6.1, windows x64

When I add ANT components to my layout without errors, after that I draw paths to connect them, then I attempt to convert path of (device layer full etch) to waveguide [W or 2], I receive this error message:

'Waveguide' is not available. Check that the library was loaded successfully.
  C:\Users\Hasan\KLayout\salt\siepic_tools\python\SiEPIC\scripts.py:686
  C:/Users/Hasan/KLayout/salt/siepic_tools/pymacros/Keybindings/Waveguides/Path to Waveguide - Hotkey.lym:5

For the same layout, if I change the technology to AMF, add two random components, then path, and then convert path to waveguide using SepicTools, it works flawlessly.

Regards

lukasc-ubc, 2022-06-26:
In which PDK does it not work?  If it is the ANT one, you will need to let them know. If EBeam, please post on the EBeam repository.

hasanhoji, 2022-06-26:
Hi Lukas,
`

It is  **NanoSOI_PDK_v6.1** the newest version which I downloaded from ANT official website "https://www.appliednt.com/nanosoi/sys/resources/pdk/"

Okay, I will send same email to ANT technical support
Regards
`

closed 2022-07-26 by mustafacc