SiEPIC/SiEPIC-Tools #174: Unable to measure waveguide length

mroddewig, 2022-07-26:
With the klayout file for Lukas Chrostowski's edx course final exam (not sure I can post) I'm unable to measure the waveguide length. I select a waveguide, click the waveguide length menu item, and then get a dialog box telling me to select a waveguide to measure. A video of the problem is here: https://montanaedu-my.sharepoint.com/:v:/g/personal/j39g462_msu_montana_edu/ESXrovVGpZJPtM--r6ULgyQBAsJ3cZ_ogpAtdMeH-ZJ57g?e=5Eqpng

I'm using the SiEPIC tools version 0.39.1 and klayout 0.27.10. Thanks!

jevillegasd, 2022-08-17:
Hey @mroddewig !

I hope this is still useful after many days. It looks to me that your waveguides are no longer associated with the SiEPIC parametric cell. I can tell by their names (with all the '$' symbols on them). It may have happened that you copied and pasted the layout from one file to another that didn't have the correct technology selected, or that you opened the file in a KLayout version without SiEPIC tools installed.

You can see an example in the files I am attaching, both have the same cell, but in one it is still the SiEPIC parametric waveguide cell, while in the other it's just a static copy of it. An easy way to test if your waveguides are correctly instantiated by  selecting them and pressing 'Q', this will open the properties window; if the 'PCell Parameters' tab is disabled, your object is static (and you cannot measure its length).

[example_cells.zip](https://github.com/SiEPIC/SiEPIC-Tools/files/9356681/example_cells.zip)

The best solution in your case is to draw the waveguides again. Here is an example of the current release of SiEPIC working fine :) (with the attached file) :

[image]

Super Secret: Even if the waveguide is now static, normally upon creation the length gets written down as a small text close to the first PIN of the waveguide, so you can zoom in that area and you might find the value there.

mroddewig, 2022-08-22:
Hi Juan,

Thanks for getting back to me! I would like to get this figured out as I want to continue using klayout beyond the course so no worries about the delay.

I opened the test_yesPCell_associated.GDS file in my klayout and got the same (width,) error as before. Is this the expected behavior? 

Thanks!
Mike

jevillegasd, 2022-08-24:
Hey Mike,

Sorry, that may be my fault because I might be using a different version for the PDK than yours. Do you mind trying to draw one waveguide on a new file and testing if it works?   Basically, just draw a path in any layer, hit 'W' on your keyboard, and then press Alt+Shift+W to measure the length. Let me know if that works.

If it does, the error I simply that the file you are checking used a different version of the EBeam PDK or SiEPIC Tools and now there is some property in the middle ('width') that changed names, and it is giving you trouble. If it doesn't, maybe you can send the file that you create and I can check if I see what's the source of the error. Probably reinstalling KLayout and SiEPIC tools does the job.

Best, Juan

lukasc-ubc, 2022-10-07:
@jaspreetj can you please address?

lukasc-ubc, 2022-10-07:
@mroddewig try upgrading to the latest siepic-tools and ebeam pdk.  the 'width' error has been fixed.

mroddewig, 2022-10-07:
Thanks, will do!

mroddewig, 2022-10-11:
Hi Juan,

On a fresh install of klayout everything worked perfectly this time. Not sure what got broken but it appears to be fixed now, so you can close the ticket. Thanks for all the help!

Mike