SiEPIC/SiEPIC-Tools #183: DC source issue when do the interconnect simulation.

xxma94, 2022-08-29:
In my SiEPIC-Tools 0.3.91, the code of find electrical IO pins: 
DCsources += "N" + str(Vn) + NetName + " dcsource amplitude=0 sch_x=%s sch_y=%s\n" % (-2-Vn/3., -2+Vn/8.) 
will raise an error in my interconnect (2022 R1), which is (1) Error loading netlist: the element 'dcsource' is not available in the library. 
The auto generated netlist will need an element called 'dcsource'. However, in lumerical the dc source is called 'DC Source'. When I manually change the 'dcsource' to 'DC Source' in netlist file (*.spi), the netlist works well.

jevillegasd, 2022-08-30:
I am using 2021-R2.3 and I see the same specification for the DC Source name. I wonder on which version of Lumerical the source used the previous naming.

lukasc-ubc, 2022-09-08:
You make a good suggestion.  To fix any issues, we would just need to check all the PDKs and make sure that they have the correct name and don’t count on the _ conversion.