SiEPIC/SiEPIC-Tools #212: XSection EBeam Tool Error

nicteb, 2024-05-27:
When trying to use the XSection tool on components from the EBeam library (v0.4.3), ANT components (v0.4.3), or SiN library (v0.4.3), the following error is displayed and no cross-section is shown:

[image]

A blank new layout tab is opened titled [XSECTION] but is blank.

Windows, KLayout v0.29.1, python packages v1.2, siepic_tools v0.5.7, siepic_ebeam_pdk v0.4.9 (all up-to-date as of time of issue).

lukasc-ubc, 2024-05-28:
Can you please upgrade SiEPIC-EBeam-PDK to v0.4.11 via the package manager?

After restarting KLayout, you should see the xsection menu with v 1.8 
[image]

Then find the file EBeam/xsect/EBeam_ANT.xs

If the v 1.8 doesn't appear, it is because there are the old xsection files remaining from the v0.4.9 PDK.  The package manager doesn't delete files, only adds them...  If so, please remove:

 - EBeam/pymacros/xsection.lym
 - EBeam/pymacros/SiEPIC_xsection.lym
 
Then restart again.  If that doesn't solve it, please let me know.

Thanks
Lukas

nicteb, 2024-05-28:
Yep, after upgrading v1.8 is available and I can successfully get the cross-section to display by running the .xs file!
The shortcut doesn't seem to work though:
[image]

But the macro file is now appearing in the menu, so that helps!
[image]

lukasc-ubc, 2024-05-28:
Indeed, I think it is a minor bug here: https://github.com/klayoutmatthias/xsection/issues/34

closed 2024-05-28 by lukasc-ubc