SiEPIC/SiEPIC-Tools #219: no result in interconnect!!!

mahdijavid, 2024-08-22:
When we simulate our design in Lumerical interconnect why sometimes we get "no result" message when we try to plot the results while DRC and LVS are passed with no errors?
[image]
[test.zip](https://github.com/user-attachments/files/16710479/test.zip)

mahdijavid, 2024-08-22:
@lukasc-ubc 
@mustafacc

mustafacc, 2024-08-22:
It's possible to get empty results while passing DRC and verification. What does the INTC circuit and GDS look like?

mahdijavid, 2024-08-22:
Ii already attached a photo of intc circuit and my gds file

mustafacc, 2024-08-22:
Looks like the input source polarization is TE while the GC is TM.

mahdijavid, 2024-08-22:
I tried to make all TE
But problem is not solved
[TEST2.zip](https://github.com/user-attachments/files/16716015/TEST2.zip)
[image]
[image]

mahdijavid, 2024-08-25:
@mustafacc

Samuel-J-S, 2024-11-15:
I receive the same result with the error stating "GC_TE1550_thickness=220 deltaw=0.txt.npz failed". After opening the txt.npz file I noticed that the syntax used is compatible with Linux, not Windows. Have you solved this yet?

Samuel-J-S, 2024-11-15:
Dear @lukasc-ubc , are you familiar with this problem on Windows platforms and do you suggest running the SiEPIC Tools on Linux?

Samuel-J-S, 2024-11-18:
@mahdijavid have you solved this problem?