SiEPIC/SiEPIC-Tools #240: Question About Unexpected Loss in Lumerical INTERCONNECT Simulation

mahdijavid, 2025-02-25:
I designed a simple circuit consisting of two grating couplers connected by a waveguide. In Lumerical simulation To analyze the time delay, I added oscilloscopes at the input and output of the circuit. While I can observe the expected delay in the simulation, the signal experiences a significant loss that seems unrealistic to me. I’m concerned this might not align with the physical behavior I’d expect from such a setup.  
Could you please advise on what might be causing this excessive loss? I suspect it could be related to the component parameters, simulation settings, or perhaps an oversight in how I’ve configured the oscilloscopes or elements. Any guidance or suggestions you could offer would be incredibly helpful. I’d be happy to provide more details about my setup if needed.  
Thank you very much for your time and expertise.
Best regards,

[image]

[test002.zip](https://github.com/user-attachments/files/18969003/test002.zip)

mahdijavid, 2025-02-25:
@mustafacc @lukasc-ubc

lukasc-ubc, 2025-02-25:
Hi @mahdijavid 

Could you please plot the transmission spectrum using an ONA?  Then based on this, calculate what you would expect to see in the time domain.